RoboStack/ros-humble #269: Can't do ros2 topic list without setting ROS_DOMAIN_ID=1

bchax, 2025-03-24:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hi. As the title says, when launching an empty gazebo world and doing ros2 topic list in another window I get no response until timeout. The only workaround I found was doing ROS_DOMAIN_ID=1, it did not even work with a complete reinstall of all the packages. What could be the root problem?

Thanks in advance.

### Installed packages

```shell
(rosenv) bauti@Bautis-MacBook-Air ~ % mamba list           
# packages in environment at /Users/bauti/miniforge3/envs/rosenv:
#
# Name                    Version                   Build  Channel
abseil-cpp                20220623.0           he4e09e4_6    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
argcomplete               3.6.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
blosc                     1.21.5               h9c252e8_1    conda-forge
boost                     1.78.0          py310he53c7d2_5    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.1.0                hd74edd7_2    conda-forge
brotli-bin                1.1.0                hd74edd7_2    conda-forge
bullet                    3.21                 h69fb684_4    conda-forge
bullet-cpp                3.21            py310h2b830bf_4    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.34.4               h5505292_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2025.1.31            hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
cartographer              2.0.0           lua54hd0cfe46_12    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_1    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_1    conda-forge
cctools                   973.0.1             h4faf515_13    conda-forge
cctools_osx-arm64         973.0.1             h7a0f565_13    conda-forge
ceres-solver              2.1.0                hfaa3f39_2    conda-forge
cffi                      1.17.1          py310h497396d_0    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
clang                     16.0.1               h1e5164c_0    conda-forge
clang-16                  16.0.1          default_he7141eb_0    conda-forge
clang_osx-arm64           16.0.1               hdccda30_1    conda-forge
clangxx                   16.0.1          default_ha0f6327_0    conda-forge
clangxx_osx-arm64         16.0.1               h0c1d50e_1    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_1    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_1    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_1    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310hbe9552e_2    conda-forge
colcon-core               0.19.0             pyhd8ed1ab_1    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.3.0              pyhd8ed1ab_1    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.9              pyhff2d567_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
colcon-ros                0.5.0              pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.5.0              pyhd8ed1ab_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
[... 691 lines cut ...]
 populated config files : /Users/bauti/miniforge3/.condarc
                          /Users/bauti/.condarc
          conda version : 24.11.3
    conda-build version : not installed
         python version : 3.12.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=24.11.3=0
                          __osx=15.0=0
                          __unix=0=0
       base environment : /Users/bauti/miniforge3  (writable)
      conda av data dir : /Users/bauti/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/bauti/miniforge3/pkgs
                          /Users/bauti/.conda/pkgs
       envs directories : /Users/bauti/miniforge3/envs
                          /Users/bauti/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.11.3 requests/2.32.3 CPython/3.12.8 Darwin/24.0.0 OSX/15.0 solver/libmamba conda-libmamba-solver/24.9.0 libmambapy/1.5.12
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2025-04-08:
Hi @bchax - have you checked whether this happens on linux as well?

traversaro, 2025-04-10:
By the way @bchax  it seems you are using quite old packages, can you try with the latest version of the packages? Thanks!

Tobias-Fischer, 2025-07-23:
Please let us know if this still happens with the newest packages; closing in the meantime.

closed 2025-07-23 by Tobias-Fischer