RoboStack/ros-humble #271: feat: Autocompletion on pixi

EliaCereda, 2025-03-27:
Pixi has implemented support for shell autocompletion scripts a few weeks ago in (https://github.com/prefix-dev/pixi/pull/3219 and https://github.com/prefix-dev/pixi/issues/2366)

To take advantage of this feature, a package needs to install autocompletion files in one of these directories depending on the user's shell:
- `share/bash-completion/completions`
- `share/zsh/site-functions`
- `share/fish/vendor_completions.d`

I tested this with both Bash and Zsh on macOS, using the attached scripts (unzip in the root of your pixi workspace) and it's working great. What's missing is just RoboStack installing these files automatically as part of the distribution. 

Is this the correct repo to discuss how to possibly get this merged? We encountered this as part of a university course on ROS we are teaching, so we'd really benefit from a fix. I can send a pull request after we define how you'd like to approach this.

Thank you!

[ros-autocompletion.zip](https://github.com/user-attachments/files/19483576/ros-autocompletion.zip)

Tobias-Fischer, 2025-03-27:
A question for @wolfv I think how to best achieve this

Tobias-Fischer, 2025-03-27:
PS: which uni / course do you teach?

traversaro, 2025-03-27:
I guess also @ruben-arts may be interested.

EliaCereda, 2025-03-27:
It's Robotics in the Master in AI at USI (Lugano, Switzerland), https://search.usi.ch/en/courses/35270726/robotics 

Thanks to Pixi we've been able to migrate from a cumbersome VM that we used to prepare and distribute to students, to a single set of instructions that they can follow themselves to setup a native environment in macOS / Ubuntu / Windows. It's a great improvement!

ruben-arts, 2025-03-27:
That's great to hear @EliaCereda, exactly what we are designing pixi for!

The quickest way would be to add these scripts to the `colcon` and the package the `ros2` cli comes from. It's also pretty normal to create a separate package like `colcon-completions` which you would version similarly to `colcon`. This way, a user can decide to install the completions. 

The extra package has my preference but might not be worth the work as these environments are huge anyway.

traversaro, 2025-03-27:
I am bit confused on how this works anyhow. I have a bit of experience with feedstock providing autocomplete scripts (see ), but I am not sure how the proposed one works.

In particular if the autocomplete script is:

~~~
eval "$(register-python-argcomplete colcon)"
~~~

who installs the `register-python-argcomplete` executable? Python?

traversaro, 2025-03-27:
On the colcon side I found https://github.com/colcon/colcon-argcomplete, that is packaged in https://github.com/conda-forge/colcon-argcomplete-feedstock, have you tried if just installing `colcon-argcomplete` works fine?

traversaro, 2025-03-27:
> On the colcon side I found https://github.com/colcon/colcon-argcomplete, that is packaged in https://github.com/conda-forge/colcon-argcomplete-feedstock, have you tried if just installing `colcon-argcomplete` works fine? (in that case, a possible action is just to document that this needs to be installed for autocompletion).

By looking at the installed files ( https://conda-metadata-app.streamlit.app/?q=conda-forge%2Fnoarch%2Fcolcon-argcomplete-0.3.3-pyhd8ed1ab_1.conda ) I guess it does not work out of the box, we probably need to install some scripts in `share/bash-completion/completions` and `share/zsh/site-functions` that call the existing scripts, and verify if they do not conflict with then the same script is called as part of the colcon workspace activation.

traversaro, 2025-03-27:
See also https://github.com/colcon/colcon-argcomplete/issues/41 and https://github.com/colcon/colcon-argcomplete/issues/43 .

EliaCereda, 2025-03-31:
I just spent a bit more time looking at what ROS actually does to enable autocompletion. 

The sequence for a normal ROS installation is:
- The user sources the `setup.sh` for a given workspace
- `setup.sh` runs `source local_setup.sh` for the current workspace and all underlying workspaces
- Each `local_setup.sh` runs `_local_setup_util.py` to collect the activation commands of all installed packages
- Activation commands are loaded from `share/<package name/package.dsv` if it exists, otherwise default to `share/<package name/local_setup.{sh,bash,zsh}`
- `local_setup.sh` evals the output of `_local_setup_util.py` to run the commands inside the user's shell

Here's the output I get from `_local_setup_util.py` on my system:
```sh
AMENT_CURRENT_PREFIX="/<path>/.pixi/envs/default" _ament_prefix_sh_source_script "/<path>/.pixi/envs/default/share/ros2cli/environment/ros2-argcomplete.zsh"
AMENT_CURRENT_PREFIX="/<path>/.pixi/envs/default" _ament_prefix_sh_source_script "/<path>/.pixi/envs/default/share/ros_environment/environment/0.ros_distro_check.sh"
export ROS_DISTRO="humble"
# export ROS_LOCALHOST_ONLY="0"
export ROS_PYTHON_VERSION="3"
export ROS_VERSION="2"
AMENT_CURRENT_PREFIX="/<path>/.pixi/envs/default" _ament_prefix_sh_source_script "/<path>/.pixi/envs/default/share/rosidl_cli/environment/rosidl-argcomplete.zsh"
```

_ament_prefix_sh_source_script is defined in `local_setup.sh`, it just sources the file if it exists, and ignores it otherwise.

Out of those commands, `ros2-argcomplete.sh` doesn't do anything except calling `register-python-argcomplete`:
```sh
if type register-python-argcomplete3 > /dev/null 2>&1; then
  eval "$(register-python-argcomplete3 ros2)"
elif type register-python-argcomplete > /dev/null 2>&1; then
  eval "$(register-python-argcomplete ros2)"
fi
```

So it seems to me we have two problems here:
- When launching a pixi shell, Pixi sources the `setup.sh` with its custom system that preserves environment variables but not other shell customization (such as autocompletion). This breaks the assumption of ROS packages that they'll have their activation scripts sourced directly from the user shell.
- Even if I source the `setup.sh` inside a pixi shell manually, it doesn't seem to enable the autocompletion correctly. Running `eval "$(register-python-argcomplete ros2)"` directly works instead.

The latter might be the same problem from the StackOverflow answer I linked above (I haven't investigated further)
> I haven't looked deep into how it works, the register-python-argcomplete commands are from this answer:
> https://robotics.stackexchange.com/a/100987

The former however appears to me as a fundamental problem in the way Pixi handles activation scripts. Do you have ideas for how ROS should fit in your design?

traversaro, 2025-03-31:
Thanks for the in-detail analysis. From my point of view, the simple solution for this is:
* Modify/patch the `colcon-argcomplete` conda-forge recipe to install its argcomplete script in the folders supported by pixi. In this case, also just copying the files in the `build.sh` script is doable.
* Modify/patch the `ros-humble-ros2cli` robostack package to install its argcomplete script in the folders supported by pixi. In this case, modifying the `build.sh` is more tricky as that is generated by vinca, an easier option may be to do a simple patch to the CMakeLists.txt of `ros2cli` to install them in the correct place.

My only doubt is for @ruben-arts , i.e. quoting my earlier comment:

> Just to understand, what is the reason not to just add the autocomplete script in the main package? The increase time spent in the activation scripts, or people not wanting autocomplete for some reason?

Unless there is a compelling reason do to that, I would avoid the complexity of adding an output for the recipe just for the autocomplete script, that especially for robostack packages is really far from trivial.

wolfv, 2025-03-31:
I think it's fine to add the autocompletion script to the main output. 

They will only be sourced when running `pixi shell`, so `pixi run` would not have a performance hit.

And for `pixi shell` it's a nice-to-have feature in my opinion :)

traversaro, 2025-03-31:
Ah, one additional note, probably we may want to patch:
* https://github.com/ros2/ros2cli/blob/0e34880d5965a979ef41a7a9311e0abba7bc915e/ros2cli/completion/ros2-argcomplete.bash#L15
* https://github.com/colcon/colcon-argcomplete/blob/master/completion/colcon-argcomplete.bash#L2

to simply be `eval "$(register-python-argcomplete colcon)"` and `eval "$(register-python-argcomplete ros2)`. As far as I can see, `register-python-argcomplete3` was never packaged in conda-forge, so the only case in which `register-python-argcomplete3` exists is on old Ubuntu distros, and preferring `register-python-argcomplete3` over `register-python-argcomplete` will mean that the apt python is used in place of the conda/pixi python, resulting in wrong autocompletes.

EliaCereda, 2025-03-31:
I went ahead and tried to patch `ros2cli` and `rosidl_cli` to install their autocompletion scripts in the proper locations. Is anything else needed for these patches to be picked up by the build system?

traversaro, 2025-03-31:
> I went ahead and tried to patch `ros2cli` and `rosidl_cli` to install their autocompletion scripts in the proper locations. Is anything else needed for these patches to be picked up by the build system?

Yes, but I will comment in the PR, thanks for looking into this!

traversaro, 2025-04-10:
This should have been fixed by https://github.com/prefix-dev/pixi/pull/3506, @EliaCereda did you checked if everything works as expected? Thanks!

EliaCereda, 2025-04-10:
I tried briefly yesterday but it didn't seem to work, I still need to investigate. `ros2cli` is installing the files but autocompletion is not activated, while `rosidl-cli` doesn't install the file at all.

Do you have any idea why for the latter?

traversaro, 2025-04-10:
This is interesting. Can you report a `conda list` or `pixi list` ? So we verify that indeed the build number of the packages is indeed `_9` (the version that should include our patches), see https://anaconda.org/robostack-humble/ros-humble-ros2cli/files and https://anaconda.org/robostack-humble/ros-humble-rosidl-cli/files . You can even enforce this in your pixi file if you want, with:

~~~
ros-humble-ros2cli = { version = "*", build = "*_9" }
ros-humble-rosidl-cli = { version = "*", build = "*_9" }
~~~

wolfv, 2025-04-10:
Please let me know if there is an issue in the pixi side of things :)

2lian, 2025-11-22:
Hi there!

Is this supposed to be working? because it is not working for me despite using the latest versions of the packages.

```
pixi list -e humble | grep cli

Environment: humble
jupyter_client                                     8.6.3               pyhd8ed1ab_1             103.8 KiB   conda  https://prefix.dev/conda-forge/
nbclient                                           0.10.2              pyhd8ed1ab_0             27.4 KiB    conda  https://prefix.dev/conda-forge/
prometheus_client                                  0.23.1              pyhd8ed1ab_0             53.3 KiB    conda  https://prefix.dev/conda-forge/
pulseaudio-client                                  17.0                h9a6aba3_3               733.2 KiB   conda  https://prefix.dev/conda-forge/
ros-humble-ros2cli                                 0.18.12             np126py311hbc2a38a_13    72.7 KiB    conda  https://prefix.dev/robostack-humble/
ros-humble-ros2cli-common-extensions               0.1.1               np126py311hbc2a38a_13    25.4 KiB    conda  https://prefix.dev/robostack-humble/
ros-humble-rosidl-cli                              3.1.6               np126py311hbc2a38a_13    38.1 KiB    conda  https://prefix.dev/robostack-humble/
websocket-client                                   1.9.0               pyhd8ed1ab_0             60 KiB      conda  https://prefix.dev/conda-forge/
```

What I don't exactly understand either is that when I `pixi shell` then `eval "$(register-python-argcomplete colcon); eval "$(register-python-argcomplete ros2)"`, then it works fine. However I cannot automatically source those commands in my `pixi.toml` like so:

```
[feature.humble.activation]
scripts = [
"workspace/install-humble/setup.bash",
"scripts/ros_autocomplete.bash",
]
```

Tobias-Fischer, 2026-10-18:
**Fixed** for `ros2` (and `rosidl`) in the RoboStack monorepo, [RoboStack/robostack](https://github.com/RoboStack/robostack), where all ROS 2 distributions are now built by [RoboStack/robostack#70](https://github.com/RoboStack/robostack/pull/70): the bash and zsh completions are now installed where the shells pick them up automatically, including in pixi environments. For humble, it ships with the rebuild in [RoboStack/robostack#62](https://github.com/RoboStack/robostack/pull/62), so it reaches users once that's merged and uploaded; the published `ros2cli` doesn't have it yet. `colcon` completion, also asked about here, comes from conda-forge's `colcon-argcomplete`, not RoboStack.

🤖 Claude (AI assistant), on behalf of @Tobias-Fischer

closed 2026-10-18 by Tobias-Fischer