RoboStack/ros-humble #272: source install/setup.bash, no such file or directory

adlimited, 2025-03-28:
Has anyone else encountered this issue when trying to run source install/setup.bash?

Error: /opt/miniconda3/envs/ros2/local_setup.bash:.:11: no such file or directory

traversaro, 2025-03-28:
Hello @adlimited, can you please follow the issue report template by reporting the `conda info` and `conda list` (or `mamba`/`micromamba`/`pixi` equivalents)? Thanks!

adlimited, 2025-03-28:
Hi, I'm following this tutorial from Elephant Robotics:
https://docs.elephantrobotics.com/docs/gitbook-en/12-ApplicationBaseROS/12.2-ROS2/12.2.7-rivz%E4%BB%8B%E7%BB%8D%E5%8F%8A%E4%BD%BF%E7%94%A8/

Everything works fine until I run this command:
`source install/setup.bash
`
Then I get the following error:
`/opt/miniconda3/envs/ros2/local_setup.bash:.:11: no such file or directory
`

I'm using a Conda-based ROS 2 Humble environment. Here's the output of conda list 
[ros2_humble_conda_list.txt](https://github.com/user-attachments/files/19503093/ros2_humble_conda_list.txt)

traversaro, 2025-03-28:
It seems that you mixing defaults and conda-forge dependencies, can you create an environment that just install packages from `conda-forge` and `robostack-humble` (you used `robostack-staging`, but that channel is not advertised anymore, I would suggest to switch to `robostack-humble` that has all the latest packages)?

adlimited, 2025-03-28:
Just to clarify — does this mean I should delete my current environment and create a new one from scratch, but this time only using conda-forge and robostack-humble as channels?

`conda create -n ros2 python=3.10`
`conda activate ros2`
`conda config --env --add channels conda-forge`
`conda config --env --add channels robostack-humble`
`conda config --env --remove channels defaults `
`conda install ros-humble-desktop`

Then install everything again using only these two channels (and not use defaults or robostack-staging), right?
Just want to make sure before I start fresh. Thanks!

traversaro, 2025-03-28:
Exactly, that is correct, expect for two things: 

* The latest (and with latest bugfixes) packages for ros-humble are only available for Python 3.11, not 3.10. There is any reason why you want to stick to 3.10 ? 
* There is no reason to delete your current environment, you can just create a new one with a new name.

adlimited, 2025-03-28:
I actually did try using Python 3.11 earlier, but rviz2 failed to launch and threw errors during runtime. I'm working on an M1 Pro Mac, so I figured falling back to Python 3.10 might improve compatibility. That’s why I’m currently sticking to Python 3.10.

traversaro, 2025-03-28:
Ok. In that case this is a classical [XY communication problem](https://en.wikipedia.org/wiki/XY_problem). The latest (and actually only one that for which we ca realistically do bug fixes) are the `python=3.11.*` builds. Can you please try them and report the rviz2 problems in a separate issue? Thanks!

traversaro, 2025-03-28:
It is possible that the rviz2 problems were related to the mix of `defaults` and `conda-forge`/`robostack-humble` packages.

adlimited, 2025-03-28:
I created a new environment using python=3.11 and installed everything from conda-forge and robostack-humble as suggested. However, when I try to run rviz2, I get the following error:

`libc++abi: terminating due to uncaught exception of type std::runtime_error: Environment variable 'AMENT_PREFIX_PATH' is not set or empty`

It seems like the AMENT_PREFIX_PATH is not automatically configured. Is there a recommended way to set it correctly when using Conda environments on macOS (M1 Pro)?

Thanks in advance!

Tobias-Fischer, 2025-03-28:
Again, please reopen and provide all requested information in the bug report so we can help better.

closed 2025-03-28 by Tobias-Fischer

adlimited, 2025-03-28:
Even after running source install/setup.bash, the issue still persists — rviz2 still fails to launch.
Launching rviz2 remains a separate problem.

Any suggestions on how to further debug this?

Tobias-Fischer, 2025-03-28:
Please create a new issue with a bug report including all requested information