RoboStack/ros-humble #273: Issue when sourcing install/setup.bash

adlimited, 2025-03-28:
Hi, I'm following this tutorial from Elephant Robotics:
https://docs.elephantrobotics.com/docs/gitbook-en/12-ApplicationBaseROS/12.2-ROS2/12.2.7-rivz%E4%BB%8B%E7%BB%8D%E5%8F%8A%E4%BD%BF%E7%94%A8/

Everything works fine until I run this command:
`source install/setup.bash
`
Then I get the following error:
`/opt/miniconda3/envs/ros2/local_setup.bash:.:11: no such file or directory
`

I'm using a Conda-based ROS 2 Humble environment. Here's the output of conda list 
[ros2_humble_conda_list.txt](https://github.com/user-attachments/files/19503123/ros2_humble_conda_list.txt)

traversaro, 2025-03-28:
Check the answer I gave in https://github.com/RoboStack/ros-humble/issues/272 : 

> It seems that you mixing defaults and conda-forge dependencies, can you create an environment that just install packages from `conda-forge` and `robostack-humble` (you used `robostack-staging`, but that channel is not advertised anymore, I would suggest to switch to `robostack-humble` that has all the latest packages)?

traversaro, 2025-03-28:
Duplicate of https://github.com/RoboStack/ros-humble/issues/272 .