RoboStack/ros-humble #292: Failed to subscribe ROS 2 topic between pixi and Docker container with Cyclone DDS

atinfinity, 2025-04-19:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

I failed to subscribe ROS 2 topic between pixi and Docker container with Cyclone DDS.

# Steps to reproduce

## Install pixi

```bash
curl -fsSL https://pixi.sh/install.sh | bash
```

## Create project

```bash
pixi init my_ros2_project -c robostack-humble -c conda-forge
cd my_ros2_project
```

## Install package

```bash
pixi add ros-humble-desktop
```

## Launch Docker container

I enabled `host networking` in Docker Desktop via setting. 
And, I launched Docker container.

```bash
docker run --rm -it \
           --net=host \
           --ipc=host \
           ros:humble-perception \
           bash
```

I changed to Cyclone DDS on Docker container.

```bash
apt update
apt install ros-humble-rmw-cyclonedds-cpp
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

### Pub

I published topic on pixi.

```bash
pixi run ros2 topic pub --rate 1 /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"
```

### Sub

And I tried to subscribe topic on Docker container.
But, I could not subscribe topic between pixi and Docker container.

```bash
ros2 topic echo /turtle1/cmd_vel
```

### Expected behavior

I expected the following result between pixi and Docker container.

```bash
ros2 topic list
/parameter_events
/rosout
/turtle1/cmd_vel
```

```bash
ros2 topic echo /turtle1/cmd_vel
linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8
```

## Supplementary information

I changed to Fast DDS. As a result, I could subscribe topic.

```bash
ros2 topic echo /turtle1/cmd_vel
linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8
```

### Installed packages

```shell
The result of `pixi list` longer than 65536 characters.
So, I upload to <https://gist.github.com/atinfinity/77a5273feef1bfa1e48fe6ddb1d4cc3c>
```

### Environment info

```shell
$ pixi info
System
------------
       Pixi version: 0.45.0
           Platform: osx-arm64
   Virtual packages: __unix=0=0
                   : __osx=14.4=0
                   : __archspec=1=m1
          Cache dir: /Users/dandelion/Library/Caches/rattler/cache
       Auth storage: /Users/dandelion/.rattler/credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: /Users/dandelion/.pixi/bin
    Environment dir: /Users/dandelion/.pixi/envs
       Manifest dir: /Users/dandelion/.pixi/manifests/pixi-global.toml

Workspace
------------
               Name: my_ros2_project
            Version: 0.1.0
      Manifest file: /Users/dandelion/my_ros2_project/pixi.toml
       Last updated: 19-04-2025 16:28:31

Environments
------------
        Environment: default
           Features: default
           Channels: robostack-humble, conda-forge
   Dependency count: 1
       Dependencies: ros-humble-desktop
   Target platforms: osx-arm64
```

Tobias-Fischer, 2026-03-23:
Can you please try again with the latest builds @atinfinity?

atinfinity, 2026-03-28:
The result of `pixi list` is <https://gist.github.com/atinfinity/57c3994cf7338c17b4d7b76d28afb5c2>.

traversaro, 2026-03-28:
To understand if this is a problem of apt/conda interoperability or something related to your Docker setup, did you tried to also run pixi's ROS inside the Docker?  Furthermore, did you tried in explicitly setting `RMW_IMPLEMENTATION` both inside and outside the docker, testing also other RMW like `zenoh` or `fastrtps` ?

Tobias-Fischer, 2026-03-29:
That suggests that there is an issue with your Docker setup, doesn't it?