RoboStack/ros-humble #293: Package request: webots_ros2_driver

2black0, 2025-04-25:
### Package name

webots_ros2_driver

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [x] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi RoboStack maintainers,  

I would like to request the addition of the **`webots_ros2_driver`** package to the ROS Humble distribution in RoboStack. This package is critical for bridging Webots robotics simulations with ROS 2, enabling seamless integration between the two platforms.  

### **Package Overview**  
The `webots_ros2_driver` (available on [ROS Index](https://index.ros.org/p/webots_ros2_driver/)) provides essential functionalities to automate and streamline interactions between Webots and ROS 2. Key features include:  
1. **Automatic ROS 2 Interface Generation**: Creates a ROS 2 interface directly from a Webots robot model, reducing manual configuration efforts.  
2. **URDF-Based Configuration**: Allows users to define and customize the ROS 2 interface using URDF files, ensuring compatibility with existing ROS workflows.  
3. **Pluginlib Extensibility**: Supports custom plugins to extend functionality, enabling developers to tailor communication between Webots and ROS 2 nodes.  

### **Why Include This Package?**  
- **Simulation-to-ROS 2 Workflow**: Many roboticists rely on Webots for prototyping, and this package simplifies transferring simulated robots to ROS 2 environments.  
- **Community Relevance**: The package is widely used in ROS 2 projects involving simulation, and its absence limits RoboStack’s compatibility with Webots-based workflows.  
- **Plugin Ecosystem**: The plugin system encourages modular development, aligning with ROS 2’s design philosophy.  

### **Request**  
Please consider adding `webots_ros2_driver` to the RoboStack Humble repository. This inclusion would greatly benefit developers working on simulation-driven ROS 2 projects and enhance cross-platform interoperability.  

Thank you for your time and efforts in maintaining RoboStack!

### Links and references (optional)

https://index.ros.org/p/webots_ros2_driver/

wolfv, 2025-04-28:
Would you be able to send a PR? https://robostack.github.io/Contributing.html

It might be as easy as adding the package to the vinca.yaml file.

Tobias-Fischer, 2025-07-23:
See https://github.com/cyberbotics/webots_ros2/pull/733 - there needs to be some work done for osx support. @2black0 if you want to contribute, this would be a good starting point.

Tobias-Fischer, 2026-10-18:
**Still blocked upstream** on cyberbotics/webots_ros2#733. In the RoboStack monorepo, [RoboStack/robostack](https://github.com/RoboStack/robostack), where all ROS 2 distributions are now built, `webots_ros2` is selected for linux-64 only, as here, and the humble rebuild there ([RoboStack/robostack#62](https://github.com/RoboStack/robostack/pull/62)) still has 2025.0.0 and builds it only on linux-64. **Candidate for moving** there to keep tracking it: if a maintainer replies "go ahead", I'll transfer this issue there.

🤖 Claude (AI assistant), on behalf of @Tobias-Fischer