RoboStack/ros-humble #323: `ros-humble-libg2o` conflicts with `ros2-humble-camera-info-manager`

stydxm, 2025-07-06:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

```log
stydxm@aosc-32cb91 $ pixi add --feature humble ros-humble-libg2o
Error:   × failed to solve the environment
  ╰─▶ Cannot solve the request because of: The following packages are incompatible
      ├─ ros-humble-camera-info-manager >=3.1.10,<4 can be installed with any of the following options:
      │  └─ ros-humble-camera-info-manager 3.1.10 | 3.1.10 would require
      │     └─ ros2-distro-mutex >=0.6.0,<0.7.0a0, which can be installed with any of the following options:
      │        └─ ros2-distro-mutex 0.6.0
      ├─ ros-humble-desktop * cannot be installed because there are no viable options:
      │  ├─ ros-humble-desktop 0.10.0 | 0.10.0 would require
      │  │  ├─ ros2-distro-mutex 0.6.* humble_*, which can be installed with any of the following options:
      │  │  │  └─ ros2-distro-mutex 0.6.0
      │  │  └─ ros-humble-turtlesim *, which cannot be installed because there are no viable options:
      │  │     ├─ ros-humble-turtlesim 1.4.2 | 1.4.2 would require
      │  │     │  └─ qt-main >=5.15.15,<5.16.0a0, which cannot be installed because there are no viable options:
      │  │     │     └─ qt-main 5.15.15 | 5.15.15 | 5.15.15 | 5.15.15 | 5.15.15 | 5.15.15 would constrain
      │  │     │        └─ qt ==5.15.15, which conflicts with any installable versions previously reported
      │  │     ├─ ros-humble-turtlesim 1.4.2 would require
      │  │     │  └─ ros2-distro-mutex 0.1.*, which cannot be installed because there are no viable options:
      │  │     │     └─ ros2-distro-mutex 0.1.0, which conflicts with the versions reported above.
      │  │     └─ ros-humble-turtlesim 1.4.2 would require
      │  │        └─ qt >=5.12.9,<5.13.0a0, which cannot be installed because there are no viable options:
      │  │           ├─ qt 5.12.9, which conflicts with the versions reported above.
      │  │           ├─ qt 5.12.9, which conflicts with the versions reported above.
      │  │           ├─ qt 5.12.9, which conflicts with the versions reported above.
      │  │           ├─ qt 5.12.9, which conflicts with the versions reported above.
      │  │           ├─ qt 5.12.9, which conflicts with the versions reported above.
      │  │           └─ qt 5.12.9, which conflicts with the versions reported above.
      │  └─ ros-humble-desktop 0.10.0 | 0.10.0 would require
      │     └─ ros2-distro-mutex 0.1.*, which cannot be installed because there are no viable options:
      │        └─ ros2-distro-mutex 0.1.0, which conflicts with the versions reported above.
      └─ ros-humble-libg2o * cannot be installed because there are no viable options:
         ├─ ros-humble-libg2o 2020.5.29 would require
         │  ├─ ros2-distro-mutex 0.6.* humble_*, which can be installed with any of the following options:
         │  │  └─ ros2-distro-mutex 0.6.0
         │  └─ qt >=5.15.8,<5.16.0a0, which cannot be installed because there are no viable options:
         │     └─ qt 5.15.8, which conflicts with the versions reported above.
         └─ ros-humble-libg2o 2020.5.29 would require
            └─ ros2-distro-mutex 0.1.*, which cannot be installed because there are no viable options:
               └─ ros2-distro-mutex 0.1.0, which conflicts with the versions reported above.
```

### Installed packages

```shell
stydxm@aosc-32cb91 $ pixi list
Package                     Version       Build                  Size       Kind   Source
_libgcc_mutex               0.1           conda_forge            2.5 KiB    conda  https://prefix.dev/conda-forge/
_openmp_mutex               4.5           2_gnu                  23.1 KiB   conda  https://prefix.dev/conda-forge/
argcomplete                 3.6.2         pyhd8ed1ab_0           41.2 KiB   conda  https://prefix.dev/conda-forge/
asio                        1.29.0        h59595ed_0             425.3 KiB  conda  https://prefix.dev/conda-forge/
binutils                    2.44          h4852527_0             34.2 KiB   conda  https://prefix.dev/conda-forge/
binutils_impl_linux-64      2.44          h4bf12b8_0             3.4 MiB    conda  https://prefix.dev/conda-forge/
binutils_linux-64           2.44          h4852527_0             35.1 KiB   conda  https://prefix.dev/conda-forge/
bzip2                       1.0.8         h4bc722e_7             246.9 KiB  conda  https://prefix.dev/conda-forge/
c-ares                      1.34.5        hb9d3cd8_0             202 KiB    conda  https://prefix.dev/conda-forge/
c-compiler                  1.10.0        h2b85faf_0             6.5 KiB    conda  https://prefix.dev/conda-forge/
ca-certificates             2025.6.15     hbd8a1cb_0             147.5 KiB  conda  https://prefix.dev/conda-forge/
catkin_pkg                  1.0.0         pyhd8ed1ab_1           52.1 KiB   conda  https://prefix.dev/conda-forge/
ceres-solver                2.2.0         cpuh39c81a5_6          1.4 MiB    conda  https://prefix.dev/conda-forge/
cmake                       3.31.6        h74e3db0_0             19.4 MiB   conda  https://prefix.dev/conda-forge/
colcon-argcomplete          0.3.3         pyhd8ed1ab_1           18.3 KiB   conda  https://prefix.dev/conda-forge/
colcon-bash                 0.5.0         pyhd8ed1ab_1           18.6 KiB   conda  https://prefix.dev/conda-forge/
colcon-cd                   0.1.1         pyhd8ed1ab_1           16.5 KiB   conda  https://prefix.dev/conda-forge/
[... 186 lines cut ...]
           Channels: https://prefix.dev/robostack-humble, https://prefix.dev/conda-forge
   Dependency count: 18
       Dependencies: ros-humble-desktop, ros-humble-camera-info-manager, ros-humble-serial-driver, ros-humble-asio-cmake-module, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, ceres-solver, tbb, sophus, asio, fmt, libgl-devel
   Target platforms: linux-64
    Prefix location: /home/stydxm/code/qtj_vision/.pixi/envs/humble
              Tasks: build

        Environment: jazzy
           Features: jazzy, build, default
           Channels: https://prefix.dev/robostack-jazzy, https://prefix.dev/conda-forge
   Dependency count: 16
       Dependencies: ros-jazzy-desktop, ros-jazzy-camera-info-manager, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, ceres-solver, tbb, sophus, asio, fmt, libgl-devel
   Target platforms: linux-64
    Prefix location: /home/stydxm/code/qtj_vision/.pixi/envs/jazzy
              Tasks: build

        Environment: kilted
           Features: kilted, build, default
           Channels: https://prefix.dev/robostack-kilted, https://prefix.dev/conda-forge
   Dependency count: 15
       Dependencies: ros-kilted-desktop, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, ceres-solver, tbb, sophus, asio, fmt, libgl-devel
   Target platforms: linux-64
    Prefix location: /home/stydxm/code/qtj_vision/.pixi/envs/kilted
              Tasks: build
```

traversaro, 2025-07-06:
Ok, I reproduced the issue with a fresh environment, it seems that the incompatibility is not between `ros-humble-libg2o` and `ros-humble-camera-info-manager`, see : 

~~~
traversaro@IITBMP014LW012:~$ conda create -n testhu -c conda-forge -c robostack-humble ros-humble-libg2o ros2-distro-mutex==0.6.* ros-humble-camera-info-manager
Channels:
 - conda-forge
 - robostack-humble
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
    current version: 25.3.0
    latest version: 25.5.1

Please update conda by running

    $ conda update -n base -c conda-forge conda

## Package Plan ##

  environment location: /home/traversaro/miniforge3/envs/testhu

  added / updated specs:
    - ros-humble-camera-info-manager
    - ros-humble-libg2o
    - ros2-distro-mutex=0.6

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    _x86_64-microarch-level-3  |        2_skylake           8 KB  conda-forge
    fmt-11.0.2                 |       h07f6e7f_1         193 KB  conda-forge
    gettext-0.25.1             |       h5888daf_0         525 KB  conda-forge
    gettext-tools-0.25.1       |       h5888daf_0         3.5 MB  conda-forge
    glib-2.84.2                |       h6287aef_0         595 KB  conda-forge
    gmock-1.15.2               |       ha770c72_0           7 KB  conda-forge
    gst-plugins-base-1.24.11   |       h651a532_0         2.7 MB  conda-forge
[...] follows the rest of the packages.
~~~

but actually between `ros-humble-libg2o` and `ros-humble-desktop`, and in particular `ros-humble-turtlesim`

~~~
traversaro@IITBMP014LW012:~$ conda create -n testhu -c conda-forge -c robostack-humble ros-humble-libg2o ros2-distro-mutex==0.6.* ros-humble-turtlesim
Channels:
 - conda-forge
 - robostack-humble
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package ros-humble-turtlesim-1.4.2-np126py311h2c3b307_7 requires qt-main >=5.15.15,<5.16.0a0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ ros-humble-libg2o =* * is installable with the potential options
│  ├─ ros-humble-libg2o 2020.5.29 would require
│  │  └─ qt >=5.15.8,<5.16.0a0 *, which can be installed;
│  └─ ros-humble-libg2o 2020.5.29 would require
│     └─ ros2-distro-mutex =0.1 *, which can be installed;
├─ ros-humble-turtlesim =* * is installable with the potential options
│  ├─ ros-humble-turtlesim 1.4.2 would require
│  │  └─ qt-main >=5.15.15,<5.16.0a0 *, which requires
│  │     └─ qt ==5.15.15 *, which conflicts with any installable versions previously reported;
│  └─ ros-humble-turtlesim 1.4.2 would require
│     └─ ros2-distro-mutex =0.1 *, which can be installed;
└─ ros2-distro-mutex =0.6 * is not installable because it conflicts with any installable versions previously reported.
~~~

Why this is happening? Somehow `libg2o` depends on `qt` instead of `qt-main` (the difference between the two is the presence or not of qtwebengine, we should actually double check if libg2o actually requires qtwebengine), but turtlesim requires a fairly recent qt-main, that itself requires a fairly recent qt (including qtwebengine), that is not available in conda-forge.

Possible fix:
* related libg2o dependency on qt instead of qt-main
* Ensure qt 5.15.15 is released in conda-forge

traversaro, 2025-07-06:
> Ensure qt 5.15.15 is released in conda-forge

I concentrated on this one. Indeed, looking at the package in https://anaconda.org/conda-forge/qt/files it is available on all platforms except `linux-64`, so something is going on.

traversaro, 2025-07-06:
There was a post-PR flaky network failure in https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=1244808&view=logs&jobId=7b6f2c87-f3a7-5133-8d84-7c03a75d9dfc&j=7b6f2c87-f3a7-5133-8d84-7c03a75d9dfc&t=9eb77fd2-8ddd-5444-8fc0-71cb28dcb736 : 

~~~

Traceback (most recent call last):
  File "/opt/conda/bin/conda-build", line 11, in <module>
    sys.exit(execute())
             ^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/conda_build/cli/main_build.py", line 622, in execute
    api.build(
  File "/opt/conda/lib/python3.12/site-packages/conda_build/api.py", line 211, in build
    return build_tree(
           ^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/conda_build/build.py", line 3679, in build_tree
    test(pkg, config=metadata.config.copy(), stats=stats)
  File "/opt/conda/lib/python3.12/site-packages/conda_build/build.py", line 3431, in test
    environ.create_env(
  File "/opt/conda/lib/python3.12/site-packages/conda_build/environ.py", line 1053, in create_env
    _execute_actions(prefix, precs)
  File "/opt/conda/lib/python3.12/site-packages/conda_build/environ.py", line 1341, in _execute_actions
    progressive_fetch_extract.execute()
  File "/opt/conda/lib/python3.12/site-packages/conda/core/package_cache_data.py", line 913, in execute
    raise CondaMultiError(not_cancelled)
conda.CondaMultiError: ('Connection broken: IncompleteRead(30701908 bytes read, 9334382 more expected)', IncompleteRead(30701908 bytes read, 9334382 more expected))
('Connection broken: IncompleteRead(30701908 bytes read, 9334382 more expected)', IncompleteRead(30701908 bytes read, 9334382 more expected))
~~~

restarting the CI should fix the error.

stydxm, 2025-07-07:
this seems caused other conflicts

```log
Error:   × failed to solve the environment
  ╰─▶ Cannot solve the request because of: The following packages are incompatible
      ├─ ros-humble-libg2o * can be installed with any of the following options:
      │  └─ ros-humble-libg2o 2020.5.29 would require
      │     └─ ros2-distro-mutex 0.1.*, which can be installed with any of the following options:
      │        └─ ros2-distro-mutex 0.1.0
      ├─ ros-humble-desktop * can be installed with any of the following options:
      │  ├─ ros-humble-desktop 0.10.0 would require
      │  │  ├─ numpy >=1.26.4,<2.0a0, which can be installed with any of the following options:
      │  │  │  ├─ numpy 1.26.4 would require
      │  │  │  │  └─ python >=3.10,<3.11.0a0, which can be installed with any of the following options:
      │  │  │  │     └─ python 3.10.0 | 3.10.0 | 3.10.0 | 3.10.0 | 3.10.0 | 3.10.0 | 3.10.1 | 3.10.1 | 3.10.1 | 3.10.1 | 3.10.1 | 3.10.1 | 3.10.2 | 3.10.2 | 3.10.2 | 3.10.2 | 3.10.2 |
      3.10.2 | 3.10.2 | 3.10.2 | 3.10.2 | 3.10.2 | 3.10.4 | 3.10.4 | 3.10.5 | 3.10.5 | 3.10.6 | 3.10.6 | 3.10.7 | 3.10.7 | 3.10.8 | 3.10.8 | 3.10.8 | 3.10.9 | 3.10.10 | 3.10.11 | 3.10.12
      | 3.10.13 | 3.10.13 | 3.10.14 | 3.10.15 | 3.10.15 | 3.10.15 | 3.10.16 | 3.10.16 | 3.10.16 | 3.10.17 | 3.10.18
      │  │  │  └─ numpy 1.26.4 | 1.26.4 would require
      │  │  │     └─ python >=3.9,<3.10.0a0, which can be installed with any of the following options:
      │  │  │        └─ python 3.9.0 | 3.9.0 | 3.9.0 | 3.9.0 | 3.9.0 | 3.9.1 | 3.9.1 | 3.9.1 | 3.9.1 | 3.9.1 | 3.9.1 | 3.9.2 | 3.9.4 | 3.9.5 | 3.9.6 | 3.9.6 | 3.9.7 | 3.9.7 | 3.9.7 |
      3.9.7 | 3.9.7 | 3.9.9 | 3.9.9 | 3.9.10 | 3.9.10 | 3.9.10 | 3.9.10 | 3.9.10 | 3.9.10 | 3.9.10 | 3.9.12 | 3.9.12 | 3.9.12 | 3.9.12 | 3.9.12 | 3.9.13 | 3.9.13 | 3.9.14 | 3.9.14 |
      3.9.15 | 3.9.15 | 3.9.16 | 3.9.16 | 3.9.17 | 3.9.17 | 3.9.18 | 3.9.18 | 3.9.18 | 3.9.18 | 3.9.19 | 3.9.20 | 3.9.20 | 3.9.21 | 3.9.21 | 3.9.22 | 3.9.22 | 3.9.23
      │  │  ├─ ros-humble-turtlesim *, which can be installed with any of the following options:
      │  │  │  └─ ros-humble-turtlesim 1.4.2 | 1.4.2 would require
      │  │  │     └─ python_abi 3.9.*, which can be installed with any of the following options:
      │  │  │        └─ python_abi 3.9 | 3.9
      │  │  └─ ros-humble-pcl-conversions *, which can be installed with any of the following options:
      │  │     ├─ ros-humble-pcl-conversions 2.4.5 | 2.4.5 would require
      │  │     │  └─ pcl >=1.14.1,<1.14.2.0a0, which can be installed with any of the following options:
      │  │     │     └─ pcl 1.14.1 would require
      │  │     │        └─ libboost >=1.84.0,<1.85.0a0, which can be installed with any of the following options:
      │  │     │           └─ libboost 1.84.0 | 1.84.0 | 1.84.0 | 1.84.0 | 1.84.0 | 1.84.0 | 1.84.0
      │  │     └─ ros-humble-pcl-conversions 2.4.0 | 2.4.0 would require
      │  │        └─ python_abi 3.9.*, which can be installed with any of the following options:
      │  └─ ros-humble-desktop 0.10.0 | 0.10.0 would require
      │     └─ ros2-distro-mutex 0.1.*, which can be installed with any of the following options:
      │        └─ ros2-distro-mutex 0.1.0
      └─ ros-humble-camera-info-manager >=3.1.10,<4 cannot be installed because there are no viable options:
         └─ ros-humble-camera-info-manager 3.1.10 | 3.1.10 would require
            └─ ros2-distro-mutex >=0.6.0,<0.7.0a0, which cannot be installed because there are no viable options:
               ├─ ros2-distro-mutex 0.6.0, which conflicts with the versions reported above.
               └─ ros2-distro-mutex 0.6.0, which conflicts with the versions reported above.
```

traversaro, 2025-07-07:
> this seems caused other conflicts

Can you please provide the full pixi.toml ? Otherwise is tricky to reproduce this. Furthermore, did you tried to delete your `pixi.lock` and tried again?

stydxm, 2025-07-07:
> try to relax the dependency range for the ros-humble-* packages

it works for me, thanks!!!

closed 2025-07-07 by stydxm