RoboStack/ros-humble #347: Understand who should set the IGN_GAZEBO_SYSTEM_PLUGIN_PATH for ur_simulation_gz to work fine

traversaro, 2025-09-10:
See https://github.com/RoboStack/ros-humble/pull/344 .

traversaro, 2025-09-10:
Ah, interesting. `LD_LIBRARY_PATH` should definitely be avoided. If you do not set it, what is the error that you get?

traversaro, 2025-09-10:
So is setting `IGN_GAZEBO_SYSTEM_PLUGIN_PATH` is enough? I guess that ign-gazebo should automatically add $CONDA_PREFIX\lib to the search path, but apparently that is not happening. I just checked with jazzy/gz-sim, and indeed `ros2 launch ur_simulation_gz ur_sim_control.launch.py` works fine (module a different issue of a missing `filelock` dependency that I will track separately). So I guess this was solved by all the work in relocatability in https://github.com/gazebosim/gz-sim/issues/626 . If setting `IGN_GAZEBO_SYSTEM_PLUGIN_PATH` is enough, I would just set it in the ign-gazebo activation script, and call it a day (even if I guess this would required a ros-humble full rebuild, but in the meanwhile setting manually `IGN_GAZEBO_SYSTEM_PLUGIN_PATH` will probably be a sufficient workaround.

mebbaid, 2025-09-10:
yeah setting manually `$IGN_GAZEBO_SYSTEM_PLUGIN_PATH` is enough. I can close this issue if you want.

traversaro, 2025-09-10:
> yeah setting manually `$IGN_GAZEBO_SYSTEM_PLUGIN_PATH` is enough. I can close this issue if you want.

No thanks, let's keep it open until the issue is solved out of the box, thanks!