RoboStack/ros-humble #352: Package request: lbr_fri_ros2_stack

garylvov, 2025-09-29:
### Package name

lbr_fri_ros2_stack

### Operating System(s)

- [x] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi, I'd love to add support for KUKA arms in RoboStack! Thank you for the great work on this project!

I plan to try the ```vinca.yaml``` soon

### Links and references (optional)

Package Repo: https://github.com/lbr-stack/lbr_fri_ros2_stack

mhubii, 2026-01-09:
Hi All, that's really great news. Just to provide some context: the `lbr_fri_ros2_stack` is not released in `rosdistro` because it compiles against multiple robot vendor SDK versions. Please also refer ROS Discourse: https://discourse.openrobotics.org/t/help-needed-distribute-a-ros-package-against-multiple-sdk-versions/40827/2

Would RoboStack support such scenarios?

On another note, is it also possible to distribute for `Jazzy`? The `Humble` LTS does currently lack a little behind with some features.

garylvov, 2026-01-09:
> On another note, is it also possible to distribute for `Jazzy`? The `Humble` LTS does currently lack a little behind with some features.

We can add Jazzy next!

> Hi All, that's really great news. Just to provide some context: the `lbr_fri_ros2_stack` is not released in `rosdistro` because it compiles against multiple robot vendor SDK versions. Please also [...]

I'm not sure about compiling against a specific robot vendor SDK. We intend to use FRI 1.15 on our setup.I guess that the realsense packages in robostack must solve a similar problem with librrealsense. From what I remember, each realsense driver package in RoboStack is pinned against a single librealsense version.

I'll defer to @traversaro any insight would be appreciated on this front!

garylvov, 2026-01-17:
Thank you for looking into this @traversaro those steps really help!!

We will try it! If we can't figure it out, for now, we will package only FRI 1.15 for Humble.

mhubii, 2026-01-17:
that'd be a great step! Thanks @garylvov :)

mhubii, 2026-01-19:
On top, KUKA has no remote software updates in place, complicating upgrades even further and requiring engineers on site.

For some reason, KUKA keeps:

1. Breaking SDK API. This is currently handleled via pre-processor directives. There surely are alternatives to this approach.
2. Breaking compatability, even for patches, like 1.15 -> 1.16.

kenichi-maeda, 2026-01-19:
> > I'm not sure about compiling against a specific robot vendor SDK. We intend to use FRI 1.15 on our setup.I guess that the realsense packages in robostack must solve a similar problem with librrealsense. From what I remember, each realsense driver package in RoboStack is pinned against a single librealsense version.
> > I'll defer to [@traversaro](https://github.com/traversaro) any insight would be appreciated on this front!
> 
> Ok, I now looked a bit into this. So, in RoboStack we do not have any feature to build multiple version of the same ROS/robostack package in the same rebuild. If that is something of interest for you, I think the easiest way to proceed would be to add https://github.com/lbr-stack/fri.git to conda-forge, and there support all the version you are interested in via `abi_migration_branches`, see https://conda-forge.org/docs/how-to/advanced/several-versions/ . Once that is done, we can add all the version of fri you want to target in https://github.com/RoboStack/ros-humble/blob/main/conda_build_config.yaml, and then rattler-build will automatically build multiple variants of packages that use fri for each version of fri. As long as the packages that use fri are not so much, I think that would be ok to do.

I'm working on it now (https://github.com/kenichi-maeda/staged-recipes/tree/add-fri-client-sdk), and will link to the PR once it's done

kenichi-maeda, 2026-02-03:
Just for reference, https://github.com/lbr-stack/fri is now available on conda-forge.